ByteHamster/PSE #29: [Überwachungskonsole] Dauer Alarmanzeige

HansMartinA, 2016-12-05:
Wenn der Benutzer in der Überwachungskonsole über einen ausgelösten Alarm informiert wird, wie lange soll dieser angezeigt werden bzw. soll es dem Nutzer möglich sein, diesen auszuschalten oder zu ignorieren?

HansMartinA, 2016-12-06:
Für die Dauer hätte ich festgelegt: Die Alarme werden so lange als ausgelöst angezeigt, bis die Fertigungssimulation meldet, die Alarme wurden behoben.

ByteHamster, 2016-12-07:
Haben Alarme Anfang und Ende oder sind sie einzelne Events?

ByteHamster, 2016-12-08:
Alarme sind also einzelne Events und haben keine Dauer. Würde also nur Dialogbox anzeigen, die Ampel richtet sich an die Werte der synchronen Aktualisierung von Werten

HansMartinA, 2016-12-11:
Dialog ist hinzugefügt.

ByteHamster, 2016-12-11:
Kann man den issue dann schließen?

closed 2016-12-11 by HansMartinA