ByteHamster/PSE #46: Programm Icon

ByteHamster, 2016-12-07:


Dav1dK, 2016-12-07:
Womöglich etwas mit vernetzten Maschinen und Industrie 4.0?

ByteHamster, 2016-12-07:
Simulatr
OSIP: **O**PC-UA **S**imulator for **I**ndustrial **P**lants
SIPS: **S**mall **I**ndustrial **P**lant **S**imulator
SIPSO: **S**mall **I**ndustrial **P**lant **S**imulator using **O**PC-UA
SOMLI: **S**imulating **O**PC-UA **M**achine **L**ink for the **I**ndustry 4.0
SIMLO: **S**imulating **I**ndustry 4.0 **M**achine **L**inks with **O**PC-UA

[image]

ByteHamster, 2016-12-07:
Können es ja als working title "Project Gemini" nennen oder so :P

ByteHamster, 2016-12-08:
Deadline ist 9.12. 
Bitte Kommentare zu Name und Icon
Jeder bitte 1-2 Lieblings-Namen oder Icons auswählen.
Bei kreativen Ideen gerne schreiben, ansonsten wird es eins von denen hier

NikoWil, 2016-12-08:
Name: Ich bin fuer OSIP oder SIMLO.
Icon: Ich bin fuer 5, evtl noch ein Symbol in der rechten Haelfte fuer den Empfaenger?

marimbMax, 2016-12-09:
OSIP und Icon 6.

Dav1dK, 2016-12-09:
Ich bin auch für OSIP und für eine Mischung aus 8 und 11 (also z.B. das Zahnrad in die obere rechte Ecke von 8).

ByteHamster, 2016-12-09:
We have a winner!
**OSIP: OPC-UA Simulator for Industrial Plants**

Ich stimme für **Icon 6**, also wird es das. Entwürfe folgen.

closed 2016-12-09 by ByteHamster