ByteHamster/PSE #75: Trennung zwischen Programmkomponenten klarmachen

ByteHamster, 2016-12-09:
Server/Simulation
Client/Konsole

NikoWil, 2016-12-09:
Wenn ich das Gespraech heute mittag korrekt zusammenfasse:

Die Fertigungssimulation simuliert die Werte fuer die Tanks. Sie zeigt in ihrer GUI eine schematische Darstellung der Produktionsanlage. Die Produktionsanlage laesst sich ueber die Fertigungssimulation steuern. Die Fertigungssimulation setzt in den OPC UA Servern die "gemessenen" Werte.

Die OPC UA Server aggregieren mehrere Sensorwerte, die durch die Fertigungssimulation gesetzt werden. An jedem Server koennen sich 0 bis beliebig viele OPC UA Clients registrieren. Auf Anfrage erhalten die Clients Sensordaten. Beim Server registrierte Alarme werden an die Clients geschickt wo noetig.

Die OPC UA Clients sind bei je genau einem OPC UA Server registriert. Sie fragen in regelmaessigen Intervallen die Daten von ihrem jeweiligen Server ab. Sie koennen bei ihrem Server Alarme registrieren und diese dann empfangen. Die OPC UA Clients stellen der Ueberwachungskonsole Daten bereit.

Die Ueberwachungskonsole verwendet mehrere OPC UA Clients, um ueber diese und die OPC UA Server Daten von der Fertigungssimulation zu erhalten. Ueber sie werden die OPC UA Clients indirekt gesteuert. In Ihr kann der Benutzer Alarme fuer die Clients registrieren und angeben, welche Daten von den Servern abgefragt werden sollen. Die Daten und Alarme, die sie von den Clients erhaelt, stellt sie grafisch dar.

Die einzige Verbindung von Ueberwachungskonsole und Fertigungssimulation ist ueber die OPC UA Server und Clients realisiert.

NikoWil, 2016-12-09:
Ergaenzung: Die Ueberwachungskonsole hat gar keinen Einfluss auf die Fertigungssimulation.

Die Fertigungssimulation setzt immer alle Werte in den Servern.
Die Ueberwachungskonsole kann nur festlegen, welche Werte von den Servern abgefragt werden; nicht, welche Werte die Server selbst verwalten.
Die Alarme werden nur bei den Servern registriert, nicht bei der Fertigungssimulation.

ByteHamster, 2016-12-09:
> Sie fragen in regelmaessigen Intervallen die Daten von ihrem jeweiligen Server ab

Das machen sie nur nach Aufforderung. Das heißt, die Überwachungskonsole muss dem Client in regelmäßigen Abständen sagen, er soll abrufen.

> Ueber sie werden die OPC UA Clients indirekt gesteuert.

Versteh ich nicht. Ist sie die Überwachungskonsole? Dann bitte hinschreiben 

Rest ist super, so gut hätte ich das wahrscheinlich nicht in Worte fassen können

NikoWil, 2016-12-09:
> die Überwachungskonsole muss dem Client in regelmäßigen Abständen sagen, er soll abrufen
Ernsthaft? Und wieder was gelernt... Ich dachte wir koennen dem Client einfach sagen "polle alle x ms".

> Ist sie die Überwachungskonsole?
Jap. Wird dann ergaenzt.

Dann schreibe ich nach dem Schlafen die verbesserte Version :)

Dav1dK, 2016-12-10:
> > die Überwachungskonsole muss dem Client in regelmäßigen Abständen sagen, er soll abrufen
>
> Ernsthaft? Und wieder was gelernt... Ich dachte wir koennen dem Client einfach sagen "polle alle x ms".

Darum ging es doch garnicht, ByteHamster redet von der Kommunikation zwischen Client und Überwachungskonsole.

ByteHamster, 2016-12-10:
> Darum ging es doch garnicht, ByteHamster redet von der Kommunikation zwischen Client und Überwachungskonsole.

Nein, ich rede tatsächlich von der Kommunikation zwischen Überwachungskonsole und dem darunter liegenden Client. Man kann dem Client nicht sagen, dass er automatisch abrufen soll. Man muss den Client jedes mal, wenn man einen Wert haben will, einzeln anstupsen und sagen, er soll jetzt mal beim Server nachfragen

ByteHamster, 2016-12-10:
@NikoWil übernimmst du dann diesen issue hier?

NikoWil, 2016-12-10:
Jop. Wo soll der dann ins Pflichtenheft?

ByteHamster, 2016-12-10:
Funktionale Anforderungen?

NikoWil, 2016-12-10:
Das ist etwas das ich in irgendeiner Form auch gerne am Anfang des Hefts haette. Um dem Leser den Gesamtueberblick zu geben, wie die Komponenten zusammen funktionieren.

ByteHamster, 2016-12-12:
Oder zum Komponentenmodell von Max

ByteHamster, 2016-12-12:
@NikoWil Unter `2. Zielbestimmung` steht schon was ähnliches. Das dann bitte ergänzen ;) Oder soll ich das machen?

closed 2016-12-13 by ByteHamster