ByteHamster/PSE #80: MVC

NikoWil, 2016-12-09:
Ein Kriterium von PSE ist ja, dass wir das ganze als MVC Architektur umsetzen.
Wie soll das hier eigentlich geschehen?
Ich koennte mir zz hoechstens zwei MVCs vorstellen. Einmal mit Fertigungssimulation, einmal mit Ueberwachungskonsole. Die sind ja auch getrennte Programme.

Dav1dK, 2016-12-10:
Ja mehr MVC können wir, denk ich, nicht unterbringen. So macht das schon Sinn.

ByteHamster, 2016-12-10:
Natürlich, mehr als 2 MVC wäre sinnlos.

Model: Die Klasse "Tank"
View: Die Anzeige der Simulation, greift auf das Model zu
Controller: Setzt bei geänderten Einstellungen die Werte im Model, aktualisiert den Server und weist das UI an, die neuen Werte anzuzeigen

ByteHamster, 2016-12-12:
@HansMartinA Das wolltest du machen, oder?

HansMartinA, 2016-12-13:
Ja.

closed 2016-12-13 by HansMartinA