ByteHamster/PSE #96: Keine Server

HansMartinA, 2016-12-10:
Was passiert, wenn keine Server mehr für den OPC UA Client zu erreichen sind? Wie reagiert die Überwachungskonsole darauf?

ByteHamster, 2016-12-11:
Generell nicht "Keine Server". Sobald einer der Server nicht mehr erreichbar ist, würde ich eine Fehlermeldung zeigen.

Dav1dK, 2016-12-11:
Ich würde sagen, dann öffnet sich das normale Fenster mit den Verbindungseinstellungen,mit dem Unterschied, dass oben eine kleine Fehlermeldung steht.

HansMartinA, 2016-12-11:
Ich habe es nun so gemacht, dass ein Dialog mit Fehler und die Einstellungen kommen. So kann der Nutzer u. U. neue Verbindungsinformationen eintragen und erfährt, was mit der Verbindung geschehen ist.

ByteHamster, 2016-12-11:
Kann man den issue dann schließen?

closed 2016-12-11 by HansMartinA