ByteHamster/PSE #171: Benennungen angleichen

Dav1dK, 2017-01-20:
Die Benennung von Begriffen wie Mischtank/Mixtank oder Client/Clients sollte Dokumentenübergreifend und im Dokument konsistent sein. Die Ausnahme sind Namen von Klassen, die Englisch geschrieben werden.

Dav1dK, 2017-01-20:
"typische Aktionen in OSIP" (Einleitung zu den Sequenzdiagrammen) umformulieren, um herauszustellen, dass das die komplexen Abläufe sind (das Kernstück der Anwendung)

Dav1dK, 2017-01-20:
Kontrollieren, dass wir überall von "mehreren Wrappern" statt von "einem Wrapper" sprechen. (Bezieht sich auf die OPC UA Wrapper)

NikoWil, 2017-01-21:
An einigen Stellen haben wir in den Kommentaren im Klassendiagramm Kurzformen wie "it's" statt "it is" verwendet.

In einem formalen/ technischen Dokument sollten wir nicht die umgangssprachliche Variante nehmen. Ich hab schon einige umgewandelt, bin aber nicht sicher ob ich alle gefunden habe.

ByteHamster, 2017-01-21:
Oft genug steht auch `it's` da, obwohl es `its` heißen sollte...

ByteHamster, 2017-01-21:
> An einigen Stellen haben wir in den Kommentaren im Klassendiagramm Kurzformen wie "it's" statt "it is" verwendet.

Done. Sind definitiv keine mehr drin (grep aufs Diagramm)

ByteHamster, 2017-01-21:
> Kontrollieren, dass wir überall von "mehreren Wrappern" statt von "einem Wrapper" sprechen. (Bezieht sich auf die OPC UA Wrapper)

Wir haben nur einen Wrapper. Das sagt noch nichts über die Instanzen, die man dann von den Servern erstellt.

Dav1dK, 2017-01-21:
>> Kontrollieren, dass wir überall von "mehreren Wrappern" statt von "einem Wrapper" sprechen. (Bezieht sich auf die OPC UA Wrapper)
>
> Wir haben nur einen Wrapper. Das sagt noch nichts über die Instanzen, die man dann von den Servern erstellt.

Wir haben den ServerWrapper und den ClientWrapper. Ich glaube darum geht es.

closed 2017-01-22 by ByteHamster