ByteHamster/PSE #186: Deviations from design in implementation

marimbMax, 2017-01-31:


ByteHamster, 2017-01-31:
Moved readScenario() from ExtendedParser to ScenarioParser

Dav1dK, 2017-02-05:
Methods of TankSelector are static

Dav1dK, 2017-02-05:
Override Color.equals() and Color.hashCode().

Dav1dK, 2017-02-05:
The description of the parameter of Color.set{Cyan,Magenta,Yellow} should state that the parameter needs to be in range 0 to 255, instead of 0 to 100.

Dav1dK, 2017-02-07:
Override Liquid.equals() and Liquid.hashCode().

Dav1dK, 2017-02-07:
Added more places which throw exceptions

ByteHamster, 2017-02-07:
Colors are using doubles to avoid always rounding down when using small simulation steps

Dav1dK, 2017-02-08:
Rename name of AbstractTank constructor (Tank => AbstractTank)

marimbMax, 2017-02-08:
Renamed setLocale to to setLocaleBundle in Translator. Also made bundle private and added getter method.

HansMartinA, 2017-02-17:
- Fehlender Klassenkommentar in AlarmVisualization.
- Neue Klasse MonitoringViewConstants als zentraler Ort für Konstanten der Überwachungskonsole, die in mehr als einer Klasse benötigt werden.
- Neue Klasse BarLayout als Oberklasse für sämtliche Visualisierungsklassen (Gauges, Color, FillLevel und Temperature) zur Zentralisierung des Layouts und des Labels. Damit entfallen die getLabel und getBar Methoden in diesen Klassen.
- BarVisualization zu GaugeVisualization umbenannt, um ggü. BarLayout besser unterscheiden zu können. Außerdem weiterer Konstruktor mit einer Pipe, falls die Gauge Werte der Pipe anzeigt.
- Neue Klasse MotorSpeedVisualization zur besseren Unterscheidung zwischen Pipes und dem Motor bei der Visualisierung und bei Updates.
- ProgressOverview-Konstruktor: Tank zu AbstractTank, um auch MixTank verwenden zu können.
- Umbennung der Temperaturalarme: overflow zu overheating und underflow zu undercooling (Ursprünglich bereits im Entwurf enthalten, durch einen Gitfehler allerdings nicht mehr in der Endfassung).
- MonitoringViewInterface show-Methode erhält ProductionSite zum Zugriff auf das Modell für Observerregistrierungen.
- MonitoringMenu: alle MenuButtons wurden zu MenuItems, da MenuButton die falsche Klasse für die Menüs sind.
- LoggingConsole: erbt direkt von TextArea, da dies bereits ScrollBars beinhaltet.
- Ampel: beobachtet nun AlarmVisualizations und wird bei Änderungen benachrichtigt. Ampel verwaltet dabei Listen, in welchem Zustand die Alarme sind und passt sich entsprechend an. Beobachtet damit nicht mehr direkt die Alarme aus dem Modell.
- AlarmVisualizations haben intern Referenzen auf die Alarme, die sie beobachten.

ByteHamster, 2017-02-21:
- Subscription intervall von -1, um sofort einmal zu lesen
- Settings haben parameter, der default wert übergibt

Dav1dK, 2017-02-21:
- Add reset() method to Tank and Pipe
- add initial color to TankSelector
- add SimulationViewInterface.setOverflowClosedHandler()

Dav1dK, 2017-02-21:
- Add second constructor to Pipe to set the default threshold

ByteHamster, 2017-02-21:
- Niko hat eine MenuBar klasse erstellt

ByteHamster, 2017-02-22:
interfaces wegen zyklischer abhängigkeit

Dav1dK, 2017-02-23:
Interface der Simulations GUI überarbeitet, in dem mehr Lambdas und Javafx Klassen und weniger eigene Klassen verwendet wurden.
Add "AbstractTank ProductionSite.getTank(TankSelector tank) "

Dav1dK, 2017-02-23:
Add TankAlarm.getTank()

NikoWil, 2017-02-28:
- AbstractTankDrawer drawSensors erhält Parameter double time, um bei sich bewegenden Elementen das Ausmaß der Bewegung zu bestimmen
- AbstractTankDrawer erhält Methoden getPipeStartPoint() (mittig unten am tatsächlichen Tank), getPipeEndPoint() (gibt einen aus der Reihe von Punkten die oben in den Tank führen), getPipeTopEntry() (mittig oben im Tankabschnitt für Pipes die die Simulation betreten), getPipeBotExit() (mittig unten im Tankabschnitt für Pipes die die Simulation verlassen)
- Methoden get/ set height/ width im AbstractTankDrawer entfernt da nie benötigt
- Drawer.draw erhält Parameter double time, um bei sich bewegenden Elementen das Ausmaß der Bewegung zu bestimmen
- FillSensorDrawer erhält im Konstruktor noch Parameter rows, cols als Anzahl der Reihen/ Spalten in der GUI, um relative Größen zu berechnen
- MixTankDrawer erhält im Konstruktor noch Parameter rows, cols als Anzahl der Reihen/ Spalten in der GUI, um relative Größen zu berechnen
- MotorDrawer erhält im Konstruktor noch Parameter cols als Anzahl der Spalten in der GUI, um relative Größen zu berechnen
- PipeDrawer erhält im Konstruktor noch Parameter rows als Anzahl der Reihen in der GUI, um relative Größen zu berechnen
- PipeDrawer erhält Parameter Pipe pipe, als zugehörige Pipe
- RotatingControlDrawer bekommt Methoden int getSpeed() und void setSpeed(int speed)
- RotatingControlDrawer erhaelt die Methoden rotateX und rotateY, um x bzw. y Koordinate eines Punktes um einen Winkel um einen anderen Punkt zu drehen (benutzt bei den Valves und dem Motor um das Kreuz zu drehen)
- Die Klasse SimulationMenu wurde erstellt, um die MenuBar oben im Fenster darzustellen und auszulagern.
- TankDrawer erhält im Konstruktor noch Parameter rows, cols als Anzahl der Reihen/ Spalten in der GUI, um relative Größen zu berechnen
- TemperatureSensorDrawer erhält im Konstruktor noch Parameter rows, cols als Anzahl der Reihen/ Spalten in der GUI, um relative Größen zu berechnen
- ValveDrawer erhält im Konstruktor noch Parameter rows als Anzahl der Reihen in der GUI, um relative Größen zu berechnen
- Klasse ViewConstants erstellt, die Konstanten zur Konfiguration der Darstellung der verschiedenen GUI-Elemente enthält
- die  set...Handler Methoden im SimulationMainWindow erhalten statt xyzHandler AbstractxyzHandler als Parameter
- Die Menüelemente About und Help wurden unter dem Menü ? zusammengeführt
- setX und setY im Point2D wurden entfernt da nie benutzt

- AbstractTankTab erhält im Konstruktor Parameter Tank und String name
- AbstractTankTab erhält Methoden getOutFlowSlider
- AbstractTankTab erhält Methode getGridPane(), damit Unterklassen auf die Gridpane in der das Layout festgelegt wird zugreifen können
- AbstractTankTab verliert die Methode getOutFlow, da das Listener-Modell geändert wurde
- TankTab erhält im Konstruktor Parameter Tank und String name
- TankTab verliert Methode getInFlow und getTemperaure und erhält dafür getInFlowSlider und getTemperatureSlider da das Listener Modell geändert wurde
- MixTankTab erhält im Konstruktor Parameter Tank und String name
- MixTankTab verliert Methode getMotorRPM und erhält dafür getMotorSlider da das Listener Modell geändert wurde
- SimulationControlWindow verliert Methoden getInFLow, getOutFlow, getTemperature, getMotorSpeed, setFlowRateListener wurde in setValveListener umbenannt, die setListener Methoden erhalten BiConsumer bzw. Consumer anstatt Listener als Parameter
- die Reihenfolge der Slider hat sich von inflow, outflow, temperatur zu outflow, inflow, temperatur geändert, da outflow allen Tabs gemein ist

NikoWil, 2017-03-01:
- SimulationControlWindow erhält eine Methode setControlsDisabled(), die die Slider und TextFields blockiert/ wieder freigibt, um Änderungen während eines Überlaufs zu verhindern

Dav1dK, 2017-03-01:
- Es gibt die Klasse ConfinedStringConverter

NikoWil, 2017-03-01:
- Im Menü der Simulation wurde ein Button "reset" eingefügt
- die AbstractTankTabs observen den zugewiesenen Tank, um Änderungen an Werten, die durch Szenarien verursacht werden, direkt zu übernehmen

Dav1dK, 2017-03-04:
- Die Methoden um die Szenarien in der Simulation GUI zu steuern, wurden hinzugefügt

Dav1dK, 2017-03-11:
- SimulationViewInterface.showOPCUAError()

NikoWil, 2017-03-13:
-SimulationControlInterface erhält die Methoe close(), damit beim Schließen der Simulation keine Fenster offen bleiben

Dav1dK, 2017-03-13:
- Es gibt die Klassen AlarmGroup und ObservableBoolean die der MonitoringView übergeben werden, damit die OPC UA Alarm Booleans für die Ampeln verwendet werden

ByteHamster, 2017-03-13:
Simulation: View(showOverflow), view interface(showOverflow) und overflow dialog bekommen tank statt tankSelector als Parameter

marimbMax, 2017-03-13:
Monitoring Controller: Added showDisconnectAlert and showCanNotConnectAlert to SettingsViewInterface

closed 2017-03-16 by HansMartinA