ByteHamster/PSE #206: Constants in model

ByteHamster, 2017-02-07:
- Tank size
- Maximum motor speed
- Simulation step
- Temperature min/max for displaying
- Max throughput of valve
- Current throughput of valve

ByteHamster, 2017-02-07:
Class with final public static constants

Dav1dK, 2017-02-08:
Was ist "Max throughput of valve" und "Current throughput of valve". Und gelten die Temperaturen auch für das Model, oder nur für die View?

ByteHamster, 2017-02-08:
Die Temperaturen braucht Martin. Damit er weiß, von wo bis wo die gauges darstellen sollen.

Bei valves kann es ja sein, dass der Tank drüber leer ist. Dann sind tatsächlicher Durchfluss und Ventilöffnung etwas anderes

Dav1dK, 2017-02-08:
Und warum sollte das eine Konstante sein? Das ändert sich doch ständig.

ByteHamster, 2017-02-08:
Das soll in die Pipes rein. Da passt der Titel nicht

Dav1dK, 2017-02-08:
Es gibt bei der Pipe getThreshold() das den Ventilzustand in % zurückgibt und getMaxInput(), dass zurückgibt, wieviel man momenten (also beim aktuellen Ventilzustand) reinstecken kann. Aber was tatsächlich reingesteckt wird, weiß doch eigentlich der Controller wenn er Liquid vom Tank holt. Wofür dann diese Methode?

ByteHamster, 2017-02-08:
Okay getMaxInput hatte ich übersehen

Current throughput: Damit die Gauge von Martin das anzeigen kann, das durch fließt. Sonst kann er nur die Öffnung anzeigen

Dav1dK, 2017-02-08:
Aber an welcher Stelle der Pipe? Im Durchschnitt oder am Anfang oder am Ende?

ByteHamster, 2017-02-08:
wow das wird sowieso nicht über opc ua übertragen. Dann muss Martin halt die Öffnung einfach in Prozent anzeigen.

Dav1dK, 2017-02-09:
PR: #214

commit 93df9364 (2017-02-09): [core] Add SimulationConstants

closes #206
commit 2e1a6bf5 (2017-02-09): [core] Add SimulationConstants

closes #206
commit 6ccb501c (2017-02-09): [core] Add SimulationConstants

closes #206
commit 34b395f1 (2017-02-09): [core] Add SimulationConstants

closes #206
closed 2017-02-09 by ByteHamster