ByteHamster/PSE #232: Fehler beim Zeichnen der GUI

NikoWil, 2017-02-12:
Hey, ich hatte ja schon von dem seltsamen Fehler erzaehlt, der beim Starten der GUI manchmal auftritt. Hier ist die StackTrace dazu.

Es handelt sich hier tatsaechlich um einen Error, keine Exception, Abfangen und behandeln ist also auch nicht wirklich moeglich. Ausserdem tritt der Fehler bisher ausschlieslich beim Start der GUI auf, nicht im laufenden Betrieb.

Ich habe schon Infos dazu gefunden  (http://www.jfree.org/forum/viewtopic.php?f=3&t=117577, http://stackoverflow.com/questions/14771042/javafx-runtime-thread-synchronization-error, https://github.com/Simulizer/Simulizer/issues/15). Der Fall von dort - ein anderer Thread der die UI anspricht - ist bei mir aber nicht gegeben. Es wird ausschliesslich die GUI gestartet und durch den AnimationTimer aktualisiert.

Das naechste Problem ist: der Fehler ist nicht replizierbar. Normales Ausfuehren des Codes fuehrt manchmal zum Fehler, manchmal nicht.

Hoo-ray!

Ist euch der Fehler schon einmal unter gekommen?

Ich mache den Code nochmal schoener und lade ihn die Tage mit hier hoch, damit ihr auch damit herumspielen koennt...

java.lang.InternalError: Unrecognized PGCanvas token: 68
	at com.sun.javafx.sg.prism.NGCanvas.renderStream(NGCanvas.java:1157)
	at com.sun.javafx.sg.prism.NGCanvas.renderContent(NGCanvas.java:606)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2053)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1945)
	at com.sun.javafx.sg.prism.NGGroup.renderContent(NGGroup.java:235)
	at com.sun.javafx.sg.prism.NGNode.doRender(NGNode.java:2053)
	at com.sun.javafx.sg.prism.NGNode.render(NGNode.java:1945)
	at com.sun.javafx.tk.quantum.ViewPainter.doPaint(ViewPainter.java:477)
	at com.sun.javafx.tk.quantum.ViewPainter.paintImpl(ViewPainter.java:330)
	at com.sun.javafx.tk.quantum.PresentingPainter.run(PresentingPainter.java:91)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at com.sun.javafx.tk.RenderJob.run(RenderJob.java:58)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at com.sun.javafx.tk.quantum.QuantumRenderer$PipelineRunnable.run(QuantumRenderer.java:125)
	at java.lang.Thread.run(Thread.java:745)

ByteHamster, 2017-02-13:
Wenn du den Code hochlädst, kann ich mal drüber schauen

ByteHamster, 2017-02-21:
Ist bei mir nie aufgetreten. Behoben?

ByteHamster, 2017-02-21:
Okay jetzt schon. Das scheint zu passieren, wenn du auf Koordinaten zeichnest, die nicht innerhalb des Fensters liegen. Aus timing gründen zum Beispiel

ByteHamster, 2017-03-15:
Seitdem wir Javafx selbst das Canvas resizen lassen, ist der Fehler bei mir nicht mehr aufgetreten. Hat jemand andere Erfahrungen gemacht?