ByteHamster/PSE #253: Default größe monitoring main window

ByteHamster, 2017-02-21:
Ich bin immernoch nicht sehr zufrieden damit, dass das Fenster maximiert geöffnet wird. Aber okay, damit kann man sich abfinden. Was mich ein bisschen nervt ist, dass das Fenster, wenn ich die Maximierung dann aufhebe, trotzdem so groß ist wie der komplette Bildschirm.

Jedes mal wenn ich es öffne, muss ich also den Fenster-Modus-Button klicken, skalieren und es dann wieder in der Mitte vom Bildschirm platzieren.

```
primaryStage.setMaximized(true);
```
==>
```
primaryStage.setWidth(1000);
primaryStage.setHeight(700);
primaryStage.setMaximized(true);
```
Wenn man dann den Button für den normalen Fenster-Modus klickt (oder das Fenster an der Titel-Leiste weg zieht), bekommt es dann automatisch eine praktikable Größe

ByteHamster, 2017-02-21:
Wäre das für dich ein akzeptabler Kompromiss?

HansMartinA, 2017-02-21:
Würde ich mit #244 kombinieren. Und ja, das ist akzetabel für mich.
Der Grund, dass das so hoch skaliert ist, dass JavaFX die maximale präferierte Größe für das Fenster nimmt.

closed 2017-03-04 by HansMartinA