ByteHamster/PSE #294: Control window

ByteHamster, 2017-02-28:
Window soll beim Aufruf von `show` die Werte aus dem Model übernehmen, anstatt wieder die alten anzuzeigen

NikoWil, 2017-03-01:
Die Methode zum deaktivieren ist drin (noch kein PR, da das show/ hide auch noch mit dazu kann).

Show sollte ausserhalb des Resets kein Problem sein - dann aendern sich die Werte ja nicht.
Was ich dabei dann mache guck ich mal... Geht da auch eine Methode die mit show/ hide zusammen aufgerufen wird, um es auf Model-Werte zurueck zu setzen? Je nachdem was ich sonst bei show/ hide ueberschreibe...

Dav1dK, 2017-03-01:
Ich fände es schlecht wenn man dem Kontrollfenster explizit sagen muss, die Werte zu übernehmen.
Am besten wäre es halt wenn das Kontrollfenster das Model observer würde und Änderungen selbständig mitbekommt. Ist das viel Arbeit?

Dav1dK, 2017-03-01:
Das Reset ist übrigens nicht das einzige Problem: Szenarien verändern die Werte ebenfalls. Ich denke dann ist es unumgänglich, dass das Control Window das model observed.

NikoWil, 2017-03-01:
Ja, es wird ein Observer

NikoWil, 2017-03-01:
Frage: wie kann man auf die Temperatur passend reagieren?

Observen fuehrt zz dazu, dass die Temperatur nichtmehr geaendert werden kann. Aktueller Fall: man zieht den Slider, die Temperatur im Tank aendert sich leicht, Observer (ControlWindow) wird alarmiert, rundet den Wert im Tank auf den naechsten int und setzt sich damit wieder auf den alten Wert.

Faellt nicht auf, wenn nur sehr kleine Fluessigkeitsmengen im Tank sind, da dann Aenderungen zustande kommen koennen.

ByteHamster, 2017-03-01:
Sowas hatte ich erwartet... Observer ignorieren, wenn man gerade geändert hat?

NikoWil, 2017-03-01:
Also jedes mal eine TimeStamp anlegen und im Observer damit vergleichen...?

Wuerde gehen faende ich aber sehr unschoen.

ByteHamster, 2017-03-01:
Mach dir eine Variable, die du immer aktualisierst, wenn gerade die valueChanged aufgerufen wurde. Dann im Observer nach dieser Variable filtern. Oder so ähnlich

ByteHamster, 2017-03-01:
Moment nein. Das funktioniert nicht

ByteHamster, 2017-03-01:
Wobei doch. Ist ja nicht Multi threaded... 

Also

JustSetValue=val
SetTemp(val) 

Observer 
If(Val == val) return

NikoWil, 2017-03-01:
Aber es gilt ja nicht val == val

Angenommen aktuelle Temperatur 230.
ich will auf 0 setzten

Setze also 0
Im Tank wird gemixt, es entsteht 249.9
249.9 -> 250
250 != 0
Mein Slider wird auf 250 zurueckgesetzt

Ich sehe gerade allgemein... Der Slider sollte nicht auf den aktuellen Wert reagieren... Das stellt er ja nicht dar. Also nich allgemein

NikoWil, 2017-03-01:
Oder man macht da die Temperaturen unabhaengig. Dass der Slider eben die Temperatur vom Zufluss als Abhaengigkeit hat, nicht die vom Tank wie bisher, was ja nicht so ganz passt...

ByteHamster, 2017-03-01:
Wow das ist mir gar nicht aufgefallen. Der Slider darf sich Natürlich nicht einfach nur an der Temperatur vom Tank richten... Der ist ja für die input Temperatur

NikoWil, 2017-03-01:
Wo bekomme ich die aus dem Model her?

ByteHamster, 2017-03-01:
Gibt keinen Getter ^^

NikoWil, 2017-03-01:
Klingt nach noch einem neuen Issue... Willst du oder soll ich?

ByteHamster, 2017-03-01:
Ne bau es einfach selber ein

ByteHamster, 2017-03-01:
Wie können nicht noch länger waren

NikoWil, 2017-03-01:
Ich hab es jetzt mal getInputTemperature(TankSelector selector) der ProductionSite geknuepft. Das hilft auch nicht.

closed 2017-03-01 by ByteHamster