ByteHamster/PSE #300: Drawer

ByteHamster, 2017-03-01:
Wenn der Tank drüber leer ist, sollten sich die Durchflussmesser nicht drehen

ByteHamster, 2017-03-01:
Verschieben der Regler lässt die RotatingDrawer springen

NikoWil, 2017-03-02:
Dann muss bei den RotatingDrawern wohl doch die bisherige Drehung und die Zeitdifferenz mit beruecksichtigt werden. Sonst tut er beim Verschieben des Reglers so, als haette er sich schon immer mit neuer Geschwindigkeit gedreht und hat entsprechend eine andere Position.

Kann ich heute abend rein machen.

NikoWil, 2017-03-03:
Problem: Wenn der Tank leer ist und inflow <= outflow mit inflow > 0 bleibt der Tank leer. Dennoch fliesst etwas.

Um den Fall abzufangen muesste jeder ValveDrawer mehr Teile der Simulation kennen als ich gut finde (eigene Pipe, Tank darueber, Pipe ueber dem Tank darueber)

NikoWil, 2017-03-04:
Was genau muss ich da im Liquid alles pruefen?
Ich hab mir die Klasse gerade angesehen und bin leicht verwirrt was sie tut.

ByteHamster, 2017-03-04:
> Ich hab mir die Klasse gerade angesehen und bin leicht verwirrt wie genau sich "Fluessigkeit im ersten Teil" pruefen laesst.

Die Methode ist noch nicht drin

NikoWil, 2017-03-04:
Ich weiss.
Es ging darum was geprueft werden muss, wenn man die Methode schreibt.
Nicht, dass die queue leer ist.
Dass das erste Element null ist?
Dass amount im ersten Element 0 ist?
Oder im Head?

ByteHamster, 2017-03-04:
Hmm das geht momentan tatsächlich nicht, weil es als reine queue implementiert ist. Dann müsste man wahrscheinlich eine double ended queue draus machen

Dav1dK, 2017-03-04:
Du musst prüfen ob der Amount des ersten Elements == 0 ist. Aber dazu musst du, wie ByteHamster sagte, die Queue zu einer double ended Queue machen.

closed 2017-03-08 by ByteHamster