ByteHamster/PSE #305: Prüfen ob es Sinn macht eine Untergrenze für die Größe der Fenster festzulegen

Dav1dK, 2017-03-03:


ByteHamster, 2017-03-04:
Ich bin dagegen, Untergrenzen zu definieren. Kann doch jeder skalieren wie er will... Außer man macht die Grenzen sehr klein.

NikoWil, 2017-03-04:
Passen koennte eher ein gefordertes Seitenverhaeltnis.
Dass die Simulation z.B. nicht so flach werden darf, dass der Motor oben aus den Tanks rausguckt.

Aber die Groesse selbst laesst sich an fuer sich beliebig einstellen, nur die mangelnde Aufloesung des Bildschirms auf Nutzerseite kann dabei ein Problem sein, nicht dass Dinge falsch dargestellt werden.

ByteHamster, 2017-03-04:
Ich hab glaub noch nie ein Programm benutzt, das ein bestimmtes Seitenverhältnis fordert...

NikoWil, 2017-03-04:
Nicht fest ein Seitenverhaeltnis.
Nur illegale Verhaeltnisse nicht erlauben.
Im Sinne mach es so flach wie du willst, nur nicht so weit, dass unsinnige Sachen angezeigt werden.

Aber ob es doppelt oder viermal so hoch wie breit ist ist egal, so lange es im sinnvollen Bereich bleibt.

Waere mMn hier sinnvoll. Gerade wegen des MotorDrawers.

NikoWil, 2017-03-04:
Das habe ich in der Form zwar auch noch nicht gesehen, aber wenn es passt passt es.

Besser als viele Spiele die ja direkt eine Liste an moeglichen Aufloesungen vorgeben und da noch weniger Freiheit erlauben.

ByteHamster, 2017-03-12:
Fehlt noch auf client seite

closed 2017-03-12 by HansMartinA