ByteHamster/PSE #309: [simulation-view] Interessantes Maximierungs-Verhalten.

HansMartinA, 2017-03-03:
Verschiedene Tests mit dem Nicht-Maximieren und anschließendem Maximieren brachten einen interessanten Effekt zustande beim maximierten Fenster:
[image]
Nähere empirische Untersuchungen haben ergeben:
Wird das Fenster bspw. auf folgende Größe **nicht-maximiert** gestellt:
[image]
und anschließend wieder maximiert, so sieht man am dargestellten Ausschnitt, dass die Tanks entsprechend der Fenstergröße richtig gezeichnet werden. Allerdings wird eine Größe nicht richtig gesetzt und verbleibt in der gleichen Größe wie im nicht-maximierten Größe und bildet den Ausschnitt. Der Ausschnitt ist also gleich groß dem Canvas im zuvor nicht-maximierten Fenster.
[image]
Dieses Verhalten wurde bisher nur unter Windows beobachtet. Es kommt häufiger vor, aber unter bestimmten Umständen auch wieder nicht.

NikoWil, 2017-03-03:
Interessant ist hierbei noch, dass die Größe der Drawer komplett von der Canvasgröße abhängt.
Dass die Drawer im kleinen Ausschnitt groß gezeichnet werden heißt also insbesondere, dass der Canvas mit maximiert wurde.
Dennoch ist nur ein kleiner Teil des Bildschirms als Canvas genutzt.
Der Canvas besitzt also scheinbar zwei Größen gleichzeitig.
Wie das zustande kommt ist für mich bisher nicht ersichtlich.

ByteHamster, 2017-03-04:
Hatte den Fehler unter Arch Linux / KDE auch schon

closed 2017-03-12 by HansMartinA