ByteHamster/PSE #316: Inflow regler sollte oben sein

ByteHamster, 2017-03-05:


NikoWil, 2017-03-05:
Ich wuerde die Reihenfolge so lassen.
Dann ist der Regler, der bei allen Tabs gleich ist (outflow) auch ueberall an der gleichen Stelle und ueberall flow vor Rest.

Und so herum ist die Implementierung mMn schoener, da dann in der Superklasse einfach erst alles hinzugefuegt werden kann und die Unterklassen dann ihre Sachen einfach drunter packen koennen.

ByteHamster, 2017-03-06:
Ja, die Implementierung ist definitiv so schöner. Ich hatte das aber auch schonmal gedacht und gestern hat sich dann noch Nina beschwert, dass die Reihenfolge nicht so intuitiv ist. Wäre also schon gut, wenn wir das tauschen könnten. Die Ventile sind ja auf dem Bildschirm auch anders herum angezeigt

Dav1dK, 2017-03-08:
Ich fände es anderst rum auch intuitiver.

NikoWil, 2017-03-08:
Na gut. Dann werde ich es noch aendern.

ByteHamster, 2017-03-08:
#319

closed 2017-03-08 by ByteHamster