ByteHamster/PSE #318: Versionen anpassen

Dav1dK, 2017-03-06:
Die Versionsnummer der Klassen in javadoc ist teilweise unterschiedlich, aber ich denke sie sollte überall gleich sein, oder?

ByteHamster, 2017-03-06:
Ne das bezieht sich auf die einzelne Klasse. Immer wenn man größere Änderungen an der Klasse macht, inkrementiert man die. Soweit ich das verstanden hab

NikoWil, 2017-03-06:
Wobei hier auch die Frage ist, was genau unser Nummernschema ist.
1.0 ist noch sehr klar, alles darueber hinaus haben wir - glaube ich - nie definiert.

ByteHamster, 2017-03-06:
Ich hab immer nur weiter 1.0 benutzt, Martin hat glaub immer wieder mal ein bisschen weiter gezählt (1.1, ..., 1.6). Ich halte die Nummerierung jetzt nicht für so kritisch. Lieber die anderen Bugs beheben als da Versionsnummern vereinheitlichen.

closed 2017-03-12 by ByteHamster