ByteHamster/PSE #341: Schließen

ByteHamster, 2017-03-10:
Wenn man das Hauptfenster der Simulation schließt, sollte sich das control window auch schließen

NikoWil, 2017-03-11:
Hab ich in einem branch gemacht. Settingswindow und About- und Helpdialog ebenfalls.

Ich suche gerade noch nach einer passenden Stelle um es mit dem FileDialog fuer die Szenarien zu machen.

NikoWil, 2017-03-11:
Mein Vorschlag im branch ist, dem MenuBar noch die primaryStage zu verraten und in showOpenDialog jeweils als ownerWindow die primaryStage mit zu geben.

Dann blocken alle Inputs im SimulationMainWindow bis man eine Datei ausgewaehlt/ den FileChooser geschlossen hat. Somit kann man das SimulationMainWindow nicht regulaer schliessen so lange der Dialog offen ist und damit bleibt natuerlich nie nur der Dialog offen.

Das hat ausserdem den Vorteil, dass man nichtmehr mehrere Dialoge gleichzeitig oeffnen kann was bisher noch moeglich war. Damit umgeht man naemlich, dass wir das Laden neuer Szenarien verbieten, wenn schon eins aktiv ist.

Dav1dK, 2017-03-11:
Das klingt gut finde ich, mach das.

closed 2017-03-11 by NikoWil