ByteHamster/PSE #366: Alarme werden auch beim wieder auflösen angezeigt

ByteHamster, 2017-03-11:
Meldung kommt auch, wenn gerade auf grün umgeschaltet wird

HansMartinA, 2017-03-12:
Wenn du in den Code schaust, kannst du feststellen, dass die Alarme nur ausgelöst werden, wenn sie tatsächlich ausgelöst wurden. Der doppelte Alarm bei realer Auslösung und der Alarm bei Auflösung kommen dadurch zustande, dass es Interferenzen zwischen den BooleanReceivedListenern und FloatReceivedListenern bei den Übergängen kommt.  Dies sieht man auch an den Verläufen (es gibt einen Knick): Die BooleanReceivedListener verzeichnen einen niedrigeren / höheren Wert und lösen den Alarm aus, während anschließend die FloatReceivedListener mit dem Aktualisierungsintervall einen höheren / niedrigeren Wert empfangen und dadurch den Alarm kurzzeitig deaktivieren. Dann wird mit den nächsten Aktualisierungsintervall der niedrige / hohe Werte empfangen und der Alarm nochmal ausgelöst.
Die Ursache liegt also in OPC UA oder im MonitoringController.

ByteHamster, 2017-03-12:
Dann sollten sich die Alarm Icons nur an den booleans orientieren. So war das ja auch glaub mal irgendwann mit den Alarmen geplant. Müsste halt erst der Controller fertig werden, sonst lässt sich das kaum machen

HansMartinA, 2017-03-12:
Damit bleibt der Fehler. Selbst unter Berücksichtigung der Booleans kommt es zu diesem Wechsel.

ByteHamster, 2017-03-12:
Das heißt, die booleans flackern auch?

HansMartinA, 2017-03-12:
Ja.

ByteHamster, 2017-03-12:
Gerade getestet, die reinen booleans flackern nicht. Das muss mit dem neu subscriben zusammenhängen

HansMartinA, 2017-03-12:
Taucht immer noch manchmal auf.

ByteHamster, 2017-03-12:
Jup. Hab ihn gerade gesucht

ByteHamster, 2017-03-12:
Hatte auch nichts geändert, das das beheben würde

closed 2017-03-13 by HansMartinA