ByteHamster/PSE #378: Known Bugs

Dav1dK, 2017-03-12:
- Das Menü des Monitors funktioniert manchmal nicht (man kann dann die Settings nicht mehr öffnen, passiert vor allem wenn man sich am Anfang nicht direkt verbinden kann). Vermutlich ein JavaFX Bug
- Das verbinden mit OPC UA Servern dauert lange

ByteHamster, 2017-03-12:
Das Menü funktioniert deshalb nicht, weil die disconnect fehlermeldung im Hintergrund angezeigt wird

ByteHamster, 2017-03-12:
Außerdem bitte einzelne issues, das ist übersichtlicher

Dav1dK, 2017-03-12:
Nein, es wird zu dem Zeitpunkt kein Dialog mehr angezeigt. Man kann einfach nicht auf den Knopf drücken, weil das Menü sofort wieder zu geht. Ich hab es Martin am Donnerstag gezeigt, der wusste auch keine Lösung.
Zu dem anderen Kommentar: hier will ich Bugs dokumentieren, von denen wir wissen, die wir aber nicht beheben können (also z.B. Milo oder JavaFX Bugs)

ByteHamster, 2017-03-12:
Aah okay

Dav1dK, 2017-03-15:
Der Monitor gibt manchmal eine mehrzeilige Zahlenfolge auf der Standardausgabe aus, von der wir nicht wissen woher sie kommt

Dav1dK, 2017-03-16:
Milo wirft beim beenden selten eine Exception

closed 2017-03-16 by ByteHamster