ByteHamster/PSE #413: Not on FX application thread beim Setzen der scenario infos

ByteHamster, 2017-03-15:
```
Exception in thread "Thread-15" java.lang.IllegalStateException: Not on FX application thread; currentThread = Thread-15
	at com.sun.javafx.tk.Toolkit.checkFxUserThread(Toolkit.java:236)
	at com.sun.javafx.tk.quantum.QuantumToolkit.checkFxUserThread(QuantumToolkit.java:423)
	at javafx.scene.Parent$2.onProposedChange(Parent.java:367)
	at com.sun.javafx.collections.VetoableListDecorator.setAll(VetoableListDecorator.java:113)
	at com.sun.javafx.collections.VetoableListDecorator.setAll(VetoableListDecorator.java:108)
	at com.sun.javafx.scene.control.skin.LabeledSkinBase.updateChildren(LabeledSkinBase.java:575)
	at com.sun.javafx.scene.control.skin.LabeledSkinBase.handleControlPropertyChanged(LabeledSkinBase.java:204)
	at com.sun.javafx.scene.control.skin.LabelSkin.handleControlPropertyChanged(LabelSkin.java:49)
	at com.sun.javafx.scene.control.skin.BehaviorSkinBase.lambda$registerChangeListener$0(BehaviorSkinBase.java:197)
	at com.sun.javafx.scene.control.MultiplePropertyChangeListenerHandler$1.changed(MultiplePropertyChangeListenerHandler.java:55)
	at javafx.beans.value.WeakChangeListener.changed(WeakChangeListener.java:89)
	at com.sun.javafx.binding.ExpressionHelper$SingleChange.fireValueChangedEvent(ExpressionHelper.java:182)
	at com.sun.javafx.binding.ExpressionHelper.fireValueChangedEvent(ExpressionHelper.java:81)
	at javafx.beans.property.StringPropertyBase.fireValueChangedEvent(StringPropertyBase.java:103)
	at javafx.beans.property.StringPropertyBase.markInvalid(StringPropertyBase.java:110)
	at javafx.beans.property.StringPropertyBase.set(StringPropertyBase.java:144)
	at javafx.beans.property.StringPropertyBase.set(StringPropertyBase.java:49)
	at javafx.beans.property.StringProperty.setValue(StringProperty.java:65)
	at javafx.scene.control.Labeled.setText(Labeled.java:145)
	at edu.kit.pse.osip.simulation.view.main.SimulationMainWindow.scenarioFinished(SimulationMainWindow.java:268)
	at edu.kit.pse.osip.simulation.controller.SimulationController.lambda$startScenario$12(SimulationController.java:361)
	at edu.kit.pse.osip.core.model.behavior.Scenario.run(Scenario.java:44)
	at java.lang.Thread.run(Thread.java:745)
```

NikoWil, 2017-03-15:
Wie schlimm ist der Fehler?
Laueft es einfach weiter, verhaelt es sich seltsam, stuerzt es ab?

ByteHamster, 2017-03-15:
Die Statusbar wird nicht aktualisiert, der Rest läuft weiter

HansMartinA, 2017-03-15:
Platform.runLater hilft.

commit 43ee2818 (2017-03-16): [simulation.view] Fix Issue 413.
The SimlationMainWindow was changed from outside the application thread leading to an error.
closes #413
commit b038b7be (2017-03-16): [simulation.view] Moved ui updates to jfx application thread

closes #413
commit 8a7274eb (2017-03-16): [simulation.view] Moved ui updates to jfx application thread

closes #413
commit 9b973620 (2017-03-16): [simulation.view] Moved ui updates to jfx application thread

closes #413
closed 2017-03-16 by NikoWil