ByteHamster/PSE #430: Settings der Simulation werden bei X nicht zurückgesetzt

ByteHamster, 2017-03-16:


NikoWil, 2017-03-20:
Szenario:
Im SimulationSettingsWindow werden Einstellungen geaendert.
Das SimulationSettingsWindow wird geschlossen ohne vorher auf "Speichern" gedrueckt zu haben.

Die geaenderten Einstellungen werden nicht uebernommen.

Laut Pflichtenheft ist die genaue Funktionalitaet des Knopfes hier undefiniert. Nur der Speichern- und Abbrechen-Knopf sind in ihrer FUnktion festgelegt (FA[810]).

Ich wuerde es hier dennoch fuer sinnvoll halten, die Einstellungen zu uebernehmen, wuerde dann aber noch einen extra Dialog einfuegen in der Art "Die EInstellungen wurden geaendert. Geaenderte Einstellungen uebernehmen? JA/NEIN".

Die Frage die ich mir dazu noch stelle: wann soll die Aenderung eingefuehrt werden? Denn sie betrifft erst einmal keine Abweichung vom Pflichtenheft.

Optionen die ich mir vorstellen kann:
-Jetzt, bevor wir die Szenarien testen.
-Waehrend der Szenarientests
-Nach den Szenarientests in kleinen Aenderungen die wir unabhaengig davon einfuehren
-Nach der QA Phase in der Version 1.1 mit Quality of Life Verbesserungen

Die Frage stelle ich mir vor allem deshalb, weil mMn alle Aenderungen die wir jetzt durchfuehren protokolliert werden sollten, es aber gleichzeitig gefuehlt in keine der bisherigen QA Phasen so richtig zu passen scheint.
Insofern wuerde ich noch eine Teilphase der QA Phase einfuehren, die fuer solche QoL Aenderungen vorbehalten ist.

ByteHamster, 2017-03-22:
Im Monitoring wird bei "X" einfach abgebrochen, also nicht gespeichert. Finde man braucht da keinen extra Dialog, so wenig wie man da umstellen kann. Ist ja nicht so, als dass man da 30 Minuten in den Einstellungen verbringt und es am Ende aus Versehen löscht. Wenn wirklich die Meldung rein soll, müsste auch eine Erkennung dazu, ob sich etwas geändert hat. Bei keiner Änderung dann auch keine Meldung. Das gleiche im Monitoring.

Bin dafür, den einfachen Weg zu nehmen. Einfach die reset settings funktion, die momentan beim cancel aufgerufen wird, jedes mal beim Anzeigen des Fensters aufzurufen. Dadurch hat man mit 2 Zeilen diff das Problem behoben.

HansMartinA, 2017-03-22:
Zunächst, finde ich, sollte alle Funktionalität vor den Testszenarien implementiert werden, da diese ansonsten nicht abgedeckt sind. Das bedeutet, jetzt, parallel zu den Unittests, direkt mit Unittests.
Tatsächlich würde ich auch einfach bei Anzeige der Einstellungen alles zurücksetzen. Auch wenn nicht genau definiert ist, was bei Drücken des "X" passiert, würde ich das "X" hier mit dem Abbrechen-Knopf gleichsetzen und entsprechend auch die Funktionalität.

NikoWil, 2017-03-22:
@HansMartinA Ja, mMn sollte auch die Funktionalitaet vor den Szenarien implementiert werden. Das Problem hier war ja, dass es im engeren Sinne keine geplante Funktionalitaet war.

@ByteHamster Da hab ich den Wald vor Baeumen nicht gesehen. Natuerlich kann ich es auch einfach wie den Cancel- statt wie den Save-Button behandeln... Gute Idee.

commit a581b32c (2017-03-23): [simulation.view] closes #430
commit 19c132a6 (2017-03-23): [simulation.view] closes #430
commit 560ee4a0 (2017-03-23): [simulation.view] closes #430
commit f57d80bb (2017-03-23): [simulation.view] Closing the settingsWindow now behaves like pressing the cancelButton.
closes #430
closed 2017-03-28 by NikoWil