ByteHamster/PSE #435: Alarmfenster werden mehrfach angezeigt

NikoWil, 2017-03-16:
Szenario:
Alarm A wird aktiv, A wird inaktiv, A wird wieder aktiv.

Wenn ich das alte Fenster zu Alarm A nicht geschlossen habe sind nach dem zweiten A wird aktiv zwei Fenster zu A offen.

Soll das so sein oder sollten wir tracken, ob das Fenster zu dem Alarm schon existiert?

Meine Befuerchtung ist, dass sonst ungeschickt konstruierte Szenarien sehr oft das gleiche Fenster oeffnen koennten anstatt (nur) die Konsole zu spammen.

ByteHamster, 2017-03-16:
Das hatte ich in #383 auch schon bemerkt. Mir ist aber nichts eingefallen, wie man das einigermaßen elegant beheben kann

HansMartinA, 2017-03-18:
Bei jedem Alarm wird nun mal ein Dialog angezeigt, wie es ursprünglich geplant war. Daher soll es so sein.

closed 2017-03-24 by ByteHamster